tum-vision/tum_ardrone #39: [rosrun] Couldn't find executable named drone_gui

cyliangmin, 2015-01-11:
Hello! I'm from China and I have met a problem when I used tum_ardrone. I install the ROS hydro on Ubuntu 12.04 LTS. I successfully implemented the catkin_make command, but when I run command roslaunch tum_ardrone tum_ardrone.launch, there was an error:
ERROR: cannot launch node of type [tum_ardrone/drone_stateestimation]: can't locate node [drone_stateestimation] in package [tum_ardrone]
ERROR: cannot launch node of type [tum_ardrone/drone_autopilot]: can't locate node [drone_autopilot] in package [tum_ardrone]
ERROR: cannot launch node of type [tum_ardrone/drone_gui]: can't locate node [drone_gui] in package [tum_ardrone]

And I also used the command rosrun tum_ardrone drone_stateestimation (actually I type the drone_stateestimation by myself because it didn't auto complete when I press tab-key), but these was an error like:
[rosrun] Couldn't find executable named drone_gui below /home/lmin/Documents/ros/catkin_ws/src/tum_ardrone

I have searched the answer on google but it seemed  that only few people had met this problem, so I decided to ask you! Thank you and sorry for my poor English.

cyliangmin, 2015-01-11:
Hello, I have solved the problem. The files drone_stateestimation, drone_autopilot and drone_gui had been generated under directory catkin_ws/devel/lib/tum_ardrone and should be copied to directory catkin_ws/src/tum_ardrone.

Himanshirocks, 2015-06-22:
I have stateestimation, autopilot and some UINode folders in my catkin_ws/src/tum_drone/src folder. And it still gives the above error. What is a possible solution?

cipri-tom, 2016-02-04:
@Himanshirocks the `stateestimation` you're mentioning is a folder, not a ros package.
I am facing the same issue. I believe the mentioned packages are not build, for some reason.

hil-team, 2016-02-21:
Hi, I had similar issue during the catkin_make, I was getting error
`$ /usr/bin/ld: cannot find -lglut`
so I installed glut library
`$ sudo apt-get install freeglut3-dev`
and everything worked fine

sean-duffy, 2017-03-31:
@cipri-tom Hi, I ran into a similar problem so I followed all the steps you've described but I'm still getting messages like this:

`[rosrun] Couldn't find executable named drone_stateestimation below /home/seanduffy/catkin_ws/src/tum_ardrone
`

When I run `catkin_make` it says `Built target drone_stateestimation` etc but the binaries still aren't found. I'm not sure what you mean by your last step with sourcing a file, perhaps this is my downfall?

cipri-tom, 2017-04-01:
@sean-duffy umm, IIRC, there are 2 types of 'releases' for catkin: devel and release (or something along those lines). By default, it is set up to use 'release' whereas the `drone_stateestimation` is compiled for the `devel`. So that's why you need to `source ~/catkin_ws/devel` . This will append the correct folder to your `PATH`, where the development libraries reside. To check it: `echo $PATH` and you should see that path there.

If you want this to be persistent and available in all terminals, you have add the `source` command to your `~/.bashrc` file.