tum-vision/tum_ardrone #40: Error with rosdep

mushroomgithub, 2015-01-13:
Hi,
I have this configuration:
ROS version: Indigo 1.11.9
OS version: Ubuntu 14.04 32 bits
gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1)

I follow this instructions of https://github.com/tum-vision/tum_ardrone

cd catkin_ws/src
git clone https://github.com/tum-vision/tum_ardrone.git -b indigo-devel
cd ..
rosdep install tum_ardrone
mogu@lnums:~/catkin_ws$ rosdep install tum_ardrone

ERROR: Rosdep cannot find all required resources to answer your query
Missing resource tum_ardrone
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/mogu/rosbuild_ws/package_dir
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks

can you help me ,any ideas?

mushroomgithub, 2015-01-13:
questions

buHtiGnioJ, 2016-02-12:
You should use catkin_make instead of rosdep install.

eroniki, 2016-04-29:
If you source devel/setup.bash before calling rosdep, it will find the package