tum-vision/tum_ardrone #43: Cooperative SLAM using tum_ardrone

satishvaishnav, 2015-03-03:
Hi,
I am trying to do my Master's Thesis on Cooperative SLAM using tum_ardrone package. I have made this package work for a single quadrotor and now moving further, I want to do cooperative slam wherein two quadrotors can share the map information with each other. tum_ardrone shows the map while running drone_Stateestimation node but I am trying to figure out to share this map to other quadrotor. 
Kindly give suggestions.

magcayaz, 2015-06-05:
Did you finish your thesis? Can you give us a link?

satishvaishnav, 2015-06-05:
I am still working on it. Probably after a month will be able to give you
the link.

galaxyeagle, 2015-06-11:
I am working on the same project topic, namely Cooperative SLAM using tum_ardrone. We can work together on this if you want.

magcayaz, 2015-06-11:
Cool! I can provide the ground truth data and test the algorithms with multiple ardrones as I have access to a huge Vicon Lab with a lot of ardrones.
Reach me at talha(dot)robotics@gmail.com

satishvaishnav, 2015-06-11:
That's great.

How do we start then?

magcayaz, 2015-06-14:
You can start by emailing me.

galaxyeagle, 2015-06-14:
Hey Satish, 
let us your email so that we can start our work right away. I would suggest we all join this chat group http://bit.ly/1MCfYH7 ,  which I just created. We'll continue from there.

galaxyeagle, 2015-06-14:
Oh ok ignore the chat group link I gave, its too complicated. We can chat in hangouts.

satishvaishnav, 2015-06-15:
Let's have a common chatting platform to update each other as faster as
possible.

magcayaz, 2015-06-15:
I think google hangouts is good enough. 
Please email me your google name and we will add you to our chat.

Talha