tum-vision/tum_ardrone #47: "PTAM Drone Camera Feed" not showing

souljaboy764, 2015-04-11:
I'm trying to perform state estimation on the ardrone but whenever I launch the drone_stateestimation node, I get the "PTAM Drone Map View" window but the camera feed window doesn't show. Can someone suggest some possible solutions or maybe some things that I may have overlooked?

buHtiGnioJ, 2016-02-12:
You have to start the ardrone_autonomy node as well. The state estimation node will only show the map window unless there is a video signal

pulver22, 2016-05-17:
I am facing the same situation.

Is not possible to launch PTAM Drone Camera Feed window while using tum_simulator (that doesn't require to launch ardrone_autonomy)?

souljaboy764, 2016-05-18:
You can launch it directly using tum_simulator. My problem was with my drone's camera itself and not with the package.