tum-vision/tum_ardrone #50: tum_ardrone with downward looking camera

iamfarhan007, 2015-06-02:
Hi 
Is it possible to use tum_ardrone for bottom looking drone camera.
Iv modified the parrot front looking  camra to downward looking. Apart form sending the corrent /tf what else i need to modify?

Thanks

ffurrer, 2015-06-03:
Hi,
I guess if you want some help, you should be a little more specific on what you want to achieve.

iamfarhan007, 2015-06-03:
Thanks ffurrer

I'm planning to use the ptam with downward looking camera. I will detect obsticals on the ground and build a 2d map for the turtlebot. 
I'm using parrot 2 on ubunto 14.04 RoS indigo.
In the ardrone driver.CPP  I have changed the transform of  forward camera (HD) to that of downward camera (qvga). 

Do in need to tell ptam of this rotation of camera and how can I do that?

Thanks

Farhan

iamfarhan007, 2015-07-18:
Hi
Does anyone know what in need to change to use tum_ardrone PTAM for downward looking camera?

Thanks
Farhan