tum-vision/tum_ardrone #70: Yaw causes the localization system to lose tracking

masasin, 2015-09-19:
The system is accurate when translating, but yaw causes the drone to lose all tracking, even though there are many features. It tells me that it is "attempting recovery".

Specifically, new keypoints are not added to the map.

souljaboy764, 2015-11-03:
Correct me if I'm wrong but that is a drawback of PTAM itself. Like a few other monocular SLAM systems, PTAM also performs poorly in the case of pure rotation. You need adequate translation along with rotation to prevent breaking of PTAM

pulver22, 2016-05-25:
@masasin I noticed the same behaviour and I tried to explained it here : https://github.com/tum-vision/tum_ardrone/issues/80

Did you find a solution to it?

masasin, 2016-05-27:
@pulver22 I changed the focus of my project.