tum-vision/tum_ardrone #74: Camera hangs and program crashes

ethanyu94, 2015-12-28:
When I run the program, the camera crashes after a second. I get a warning about camera calibration file not found, and drone_stateestimation crashes quickly. Sometimes, my whole computer crashes. Any advice on how I can troubleshoot this? Maybe I am building it wrong.

```
SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.16

NODES
  /
    drone_autopilot (tum_ardrone/drone_autopilot)
    drone_gui (tum_ardrone/drone_gui)
    drone_stateestimation (tum_ardrone/drone_stateestimation)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[drone_stateestimation-1]: started with pid [3062]
process[drone_autopilot-2]: started with pid [3063]
process[drone_gui-3]: started with pid [3064]
Object::connect: No such slot tum_ardrone_gui::FlatTrimClicked()
Object::connect:  (sender name:   'buttonFlattrim')
Object::connect:  (receiver name: 'tum_ardrone_guiClass')
nouveau: kernel rejected pushbuf: Invalid argument
nouveau: ch0: krec 0 pushes 1 bufs 0 relocs 0
[drone_stateestimation-1] process has died [pid 3062, exit code -11, cmd /opt/ros/jade/tum_ardrone/bin/drone_stateestimation __name:=drone_stateestimation __log:=/home/ethan/.ros/log/4764c088-ad49-11e5-a956-20c9d042f419/drone_stateestimation-1.log].
log file: /home/ethan/.ros/log/4764c088-ad49-11e5-a956-20c9d042f419/drone_stateestimation-1*.log
```

The log file says:
```
[0m[ INFO] [1451296663.823101385]: Started TUM ArDrone Stateestimation Node.[0m
set publishFreq to 30Hz
set calibFile to DEFAULT
set ts base to 1451296663
Waiting for Video
Found ARDrone Version 2
Set Camera Paramerer to: 0.771557 1.36856 0.552779 0.444056 1.15601
[0m[ INFO] [1451296664.216022236]: Video resolution: 640 x 360[0m
new delasXYZ: 40, delayVideo: 70, delayControl: 90
new delasXYZ: 40, delayVideo: 69, delayControl: 87
```

ethanyu94, 2015-12-29:
Ok, so I solved this issue by switching from the default graphics driver to the Nvidia driver. 

Now, stateestimation immediately crashes with:
```
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  155 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  27
  Current serial number in output stream:  28
[drone_stateestimation-1] process has died [pid 15144, exit code 1, cmd /home/ethan/catkin_ws/devel/lib/tum_ardrone/drone_stateestimation __name:=drone_stateestimation __log:=/home/ethan/.ros/log/0399f1dc-ae0e-11e5-a8f6-20c9d042f419/drone_stateestimation-1.log].
log file: /home/ethan/.ros/log/0399f1dc-ae0e-11e5-a8f6-20c9d042f419/drone_stateestimation-1*.log
```

The log file says:
```
[0m[ INFO] [1451381193.350452216]: Started TUM ArDrone Stateestimation Node.[0m
set publishFreq to 30Hz
set calibFile to DEFAULT
set ts base to 1451381193
Waiting for Video
Found ARDrone Version 2
Set Camera Paramerer to: 0.771557 1.36856 0.552779 0.444056 1.15601
[0m[ INFO] [1451381193.845631863]: Video resolution: 640 x 360[0m
```

ethanyu94, 2015-12-29:
Ok, just needed to restart after that driver change...

closed 2015-12-29 by ethanyu94

rubenanapu, 2019-08-02:
I had the same problem recently. I found that the reason is because `tum_ardrone/src/UINode/tum_ardrone_gui.ui` calls `FlatTrimClicked`, but on `tum_ardrone/src/UINode/tum_ardrone_gui.h` and `tum_ardrone/src/UINode/tum_ardrone_gui.cpp` the method name is `FlattrimClicked`.

Note that the `Trim` was lowercase on the header and cpp files, whereas on the .ui file it was uppercase.

After renaming `FlattrimClicked` to `FlatTrimClicked` the problem was gone.