tum-vision/tum_ardrone #78: catkin_make error for  drone_gui on Ubuntu 14.04 + ROS Indigo

azwadsabik, 2016-03-13:
Hi, 

I'm attempting to build your package using the following:
```
cd catkin_ws/src
git clone https://github.com/tum-vision/tum_ardrone.git -b indigo-devel
cd ..
rosdep install tum_ardrone
catkin_make
```
but run into the following error: 

[ 69%] Built target drone_autopilot
[ 89%] Built target drone_stateestimation
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
make[2]: *** [tum_ardrone/src/UINode/moc_RosThread.cxx] Error 1
make[1]: *** [tum_ardrone/CMakeFiles/drone_gui.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I am unsure how to approach this - any advice? 

This is on Ubuntu 14.04, and I believe I have Boost 1.54 installed. 

Any help would be appreciated.

SpaceMingebag, 2017-01-12:
Hey I'm late but I just got started messing with drones. QT and boost seems to not always play nice together. In 16.04 it gets the same error because of qt5.
Adding this to workaround your #include for all packages giving issues will work around qt5 since things are ported to qt5.

#ifndef Q_MOC_RUN
your #include files
#endif

 I added that to RosThread.cpp and tum_ardrone_gui.cpp just to be sure.

shubhraD, 2017-04-09:
@SpaceMingebag I had the same issue and your solution  worked thanks a  lot

tiralonghipol, 2017-12-04:
solved my problem too !
ros-kinetic ubuntu 16.04

Salman808, 2017-12-22:
@SpaceMinebag I have same issue, but i can't understand what you are saying?

hinduasish, 2018-08-07:
Worked for me! Great thank you so much @SpaceMingebag 
Also running ROS-KINETIC:

Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial

If anyone needs the actual RosThread file that worked for compiling message me

melhazzouri, 2019-01-04:
Hi I am trying to use tum_ardrone on ros kinetic. where do I add these two lines. Any help would be really appreciated!

20chix, 2019-02-26:
> Hi I am trying to use tum_ardrone on ros kinetic. where do I add these two lines. Any help would be really appreciated!

RosThread.h

Fred159, 2020-02-06:
Also work for me~  16.04 ubuntu and ROS Kinetic.

Fred159, 2020-02-06:
> Hey I'm late but I just got started messing with drones. QT and boost seems to not always play nice together. In 16.04 it gets the same error because of qt5.
> Adding this to workaround your #include for all packages giving issues will work around qt5 since things are ported to qt5.
> 
> Basically you are adding 2 lines.
> #ifndef Q_MOC_RUN
> #endif
> 
> It ends up looking like this in RosThread.h.
> 
> #ifndef Q_MOC_RUN
> #include "cvd/thread.h"
> #include "tum_ardrone/filter_state.h"
> #include "std_msgs/String.h"
> #include "geometry_msgs/Twist.h"
> #include "ardrone_autonomy/Navdata.h"
> #include "ros/ros.h"
> #include "sensor_msgs/Joy.h"
> #include "std_srvs/Empty.h"
> #include "std_msgs/Empty.h"
> #endif

Also work for me~ 16.04 ubuntu and ROS Kinetic.