tum-vision/tum_ardrone #82: Multiple drones

Geaper, 2016-06-01:
Hi.  I used this but my images are intercalating between each drone. How do I run 2 different tum AR drone launch for my drone1 and drone1?

Geaper, 2016-06-01:
I mean drone1 and drone2

bitelephent, 2016-11-07:
Hello Geaper，
Ruslan tell me now you are working out mutiple AR.Drones problem by using C# ,now I meet a problem ,I can control both AR.Drones and receive Navdata ,but if i receive one drone's video packet, the other is occupied , would you mind telling me how to solve it, i also serch at the fundation of Ruslan. 
THX bro.

Geaper, 2016-11-08:
I know how to solve it. I also had the same issue. Do you pay?

bitelephent, 2016-11-08:
I can pay,but i want to know whether  your program can get the frame data.In fact my program also can show the video without getting the frame data.

Geaper, 2016-11-08:
It's not easy. I was tons of hours looking at the .dlls trying to find the problem. And yes you can use 2 drones and receive both data and image without any problem. YOu just need a router and a PC.

Geaper, 2016-11-08:
I just want a game from https://www.hrkgame.com/

Geaper, 2016-11-08:
Is there a place where we can talk? I can tell you what to change in the code. Btw I can only help you if you're using C# Ar.Drone Ruslan Code.

bitelephent, 2016-11-11:
Sorry ~, the github didn't show the message . If you help me deal with it , you deserve one game !

bitelephent, 2016-11-11:
Can you use Wechat in your country , or I download the skype to contact with you .

bitelephent, 2016-11-11:
My email address is 303623227@qq.com, you can send your contact information to me. 
Thx bro!

bitelephent, 2016-11-12:
hello again bro, when u r free ,tell me which game is your wanted . Sorry for replying late. Sincerely!

Geaper, 2016-11-12:
That's a strange email. add me on facebook
https://www.facebook.com/profile.php?id=100010566253771&ref=bookmarks

bitelephent, 2016-11-12:
I couldn't open this web,which show this web didn't exist. You can search and add me by the email above.

Geaper, 2016-11-12:
What? How can facebook not exist?

bitelephent, 2016-11-12:
In fact,I have never used facebook,the facebook account is registered just now.

Geaper, 2016-11-12:
I'm sorry but I will not add you on that email. Looks very fishy to me.

Geaper, 2016-11-12:
do you have Steam?

bitelephent, 2016-11-12:
[image]

Geaper, 2016-11-12:
you are from asia? Do you have Steam / Origin?

bitelephent, 2016-11-12:
How about telling me your facebook account again,maybe it will work.

Geaper, 2016-11-12:
https://www.facebook.com/profile.php?id=100010566253771

bitelephent, 2016-11-12:
I'm downloading the messenger ,sorry.plz wait a moment.

Geaper, 2016-11-12:
ok

bitelephent, 2016-11-12:
the VPN is disconnected,so I can't use the facebook now .But don't worry,once I know how to pay for it ,i will contact you!  See u!