tum-vision/tum_ardrone #87: some errors occur 新

zhangduoshou, 2017-05-19:


zhangduoshou, 2017-05-19:
first I install ardrone_autonomy by the cmd sudo apt-get install ros-kinetic-ardrone-autonomy and it seems work well,then I install tum_ardrone by cmd as below：
cd catkin_ws/src
git clone https://github.com/tum-vision/tum_ardrone.git -b indigo-devel
cd ..
rosdep install tum_ardrone
catkin_make
it shows messages in teminal as：
Base path: /home/zhangyang/catkin_ws
Source space: /home/zhangyang/catkin_ws/src
Build space: /home/zhangyang/catkin_ws/build
Devel space: /home/zhangyang/catkin_ws/devel
Install space: /home/zhangyang/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/zhangyang/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/zhangyang/catkin_ws/build"
####
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  1%] Built target gtest
[  1%] Built target sensor_msgs_generate_messages_py
[  1%] Built target std_srvs_generate_messages_eus
[  3%] Built target jpeg_streamer
[  3%] Built target rosgraph_msgs_generate_messages_eus
[  3%] Built target roscpp_generate_messages_cpp
[  3%] Built target rosgraph_msgs_generate_messages_nodejs
[  3%] Built target geometry_msgs_generate_messages_eus
[  3%] Built target rosgraph_msgs_generate_messages_cpp
[  3%] Built target std_msgs_generate_messages_cpp
[  3%] Built target sensor_msgs_generate_messages_nodejs
[  3%] Built target geometry_msgs_generate_messages_cpp
[  3%] Built target roscpp_generate_messages_nodejs
[  3%] Built target rosgraph_msgs_generate_messages_py
[  3%] Built target roscpp_generate_messages_eus
[  3%] Built target std_msgs_generate_messages_eus
[  3%] Built target std_msgs_generate_messages_nodejs
[  3%] Built target rosgraph_msgs_generate_messages_lisp
[  3%] Built target roscpp_generate_messages_lisp
[  3%] Built target sensor_msgs_generate_messages_cpp
[  3%] Built target std_msgs_generate_messages_lisp
[  3%] Built target geometry_msgs_generate_messages_lisp
[  3%] Built target sensor_msgs_generate_messages_lisp
[  3%] Built target sensor_msgs_generate_messages_eus
[  3%] Built target topic_tools_generate_messages_nodejs
[  3%] Built target _realsense_camera_generate_messages_check_deps_SetPower
[  3%] Built target topic_tools_generate_messages_lisp
[  3%] Built target std_srvs_generate_messages_py
[  3%] Built target _realsense_camera_generate_messages_check_deps_IMUInfo
[  3%] Built target _realsense_camera_generate_messages_check_deps_CameraConfiguration
[  3%] Built target _realsense_camera_generate_messages_check_deps_IsPowered
[  3%] Built target _realsense_camera_generate_messages_check_deps_ForcePower
[  3%] Built target actionlib_generate_messages_cpp
[  3%] Built target tf_generate_messages_lisp
[  3%] Built target tf_generate_messages_eus
[  3%] Built target dynamic_reconfigure_gencfg
[  3%] Built target _realsense_camera_generate_messages_check_deps_GetIMUInfo
[  3%] Built target std_srvs_generate_messages_cpp
[  3%] Built target actionlib_generate_messages_eus
[  3%] Built target dynamic_reconfigure_generate_messages_py
[  3%] Built target dynamic_reconfigure_generate_messages_cpp
[  3%] Built target tf_generate_messages_nodejs
[  3%] Built target topic_tools_generate_messages_eus
[  3%] Built target bond_generate_messages_eus
[  3%] Built target bond_generate_messages_lisp
[  3%] Built target topic_tools_generate_messages_cpp
[  3%] Built target pcl_msgs_generate_messages_cpp
[  3%] Built target bond_generate_messages_py
[... 31 lines cut ...]
[  3%] Built target pcl_msgs_generate_messages_py
[  4%] Generating dynamic reconfigure files from cfg/r200_params.cfg: /home/zhangyang/catkin_ws/devel/include/realsense_camera/r200_paramsConfig.h /home/zhangyang/catkin_ws/devel/lib/python2.7/dist-packages/realsense_camera/cfg/r200_paramsConfig.py
[  7%] Built target realsense_camera_generate_messages_nodejs
[  7%] Built target tf_generate_messages_cpp
[ 12%] Built target realsense_camera_generate_messages_eus
[ 17%] Built target realsense_camera_generate_messages_py
[ 21%] Built target realsense_camera_generate_messages_lisp
[ 23%] Built target get_debug_info
  File "/home/zhangyang/catkin_ws/src/realsense-indigo-devel/realsense_camera/cfg/r200_params.cfg", line 32
SyntaxError: Non-ASCII character '\xe5' in file /home/zhangyang/catkin_ws/src/realsense-indigo-devel/realsense_camera/cfg/r200_params.cfg on line 32, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
realsense-indigo-devel/realsense_camera/CMakeFiles/realsense_camera_gencfg.dir/build.make:69: recipe for target '/home/zhangyang/catkin_ws/devel/include/realsense_camera/r200_paramsConfig.h' failed
make[2]: *** [/home/zhangyang/catkin_ws/devel/include/realsense_camera/r200_paramsConfig.h] Error 1
CMakeFiles/Makefile2:3195: recipe for target 'realsense-indigo-devel/realsense_camera/CMakeFiles/realsense_camera_gencfg.dir/all' failed
make[1]: *** [realsense-indigo-devel/realsense_camera/CMakeFiles/realsense_camera_gencfg.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 25%] Built target tum_ardrone_gencfg
[ 25%] Built target _tum_ardrone_generate_messages_check_deps_SetInitialReachDistance
[ 25%] Built target _tum_ardrone_generate_messages_check_deps_SetReference
[ 25%] Built target _tum_ardrone_generate_messages_check_deps_SetMaxControl
[ 25%] Built target _tum_ardrone_generate_messages_check_deps_SetStayTime
[ 25%] Built target _tum_ardrone_generate_messages_check_deps_filter_state
[ 25%] Built target _tum_ardrone_generate_messages_check_deps_SetStayWithinDistance
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

ktiwari9, 2017-06-14:
I have the same problem in ROS Kinetic. Any luck solving this ?

haochen2115, 2017-10-15:
I have the same prob, have you solved it?

rdelgadov, 2017-11-16:
I can't reproduce your problem, but i think the problem is in realsense_camera and not tum package.