tum-vision/tum_ardrone #94: Port to kinetic

rdelgadov, 2017-11-16:
Hi, 

I use the solution of #78 writen by @SpaceMingebag and the package work well in ROS Kinetic in simulator.

What I need to do to make the contribution?

lesire, 2017-11-16:
Hi,
Contributions are very welcome!! :)
Fork the repo, modify it, and create a pull request to merge it into the main repo.
Thanks !

rdelgadov, 2017-11-17:
Can you create the kinetic branch pls?

cyberphantom, 2017-11-28:
Hi just change the RosThread.h to this 

#ifndef Q_MOC_RUN
#include "cvd/thread.h"
#include "tum_ardrone/filter_state.h"
#include "std_msgs/String.h"
#include "geometry_msgs/Twist.h"
#include "ardrone_autonomy/Navdata.h"
#include "ros/ros.h"
#include "sensor_msgs/Joy.h"
#include "std_srvs/Empty.h"
#include "std_msgs/Empty.h"
#endif

as mentioned in #78

abaddriver, 2018-02-15:
hi,
I'm also trying to make the tum_ardrone work on kinetic.
for some reason, i cant get it to compile the project:
i'm following the instructions under these instructions:
"
cd catkin_ws/src
git clone https://github.com/tum-vision/tum_ardrone.git -b hydro-devel
cd ..
rosdep install tum_ardrone
catkin_make
"

the outcome is really fast and it doesn't seem to even try to compile the package.
what am i missing?

this is the output on my computer:

(cv) yoni@yoni:~/catkin_ws$ rosdep install tum_ardrone
#All required rosdeps installed successfully
(cv) yoni@yoni:~/catkin_ws$ catkin_make
Base path: /home/yoni/catkin_ws
Source space: /home/yoni/catkin_ws/src
Build space: /home/yoni/catkin_ws/build
Devel space: /home/yoni/catkin_ws/devel
Install space: /home/yoni/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/yoni/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/yoni/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _of_avihay_ronly_pkg_generate_messages_check_deps_cordinate
[ 28%] Built target of_avihay_ronly_pkg_generate_messages_lisp
[ 28%] Built target of_avihay_ronly_pkg_generate_messages_nodejs
[ 57%] Built target of_avihay_ronly_pkg_generate_messages_py
[ 85%] Built target of_avihay_ronly_pkg_generate_messages_eus
[100%] Built target of_avihay_ronly_pkg_generate_messages_cpp
[100%] Built target of_avihay_ronly_pkg_generate_messages
(cv) yoni@yoni:~/catkin_ws$

rdelgadov, 2018-02-15:
Hi,

Try clonning [this](https://github.com/rdelgadov/tum_ardrone/tree/kinetic-devel?files=1) repository in the kinetic-devel branch and do the same steps without the "-b hydro-devel"

abaddriver, 2018-02-21:
@rdelgadov i tried it, and i still get the same outcome..
it's like it doesn't know about the existence of the package